sballin/GPI_W7-X #10: feature: GUI display of last puff "fast" data

astechow, 2020-05-26:
a waveform display of both valves for the last puff at high time resolution would be nice

jameslterry, 2020-05-26:
Do you mean displaying the valve control signals or both the differential pressure signal and its derivative on a fast time base (that is set by the open duration)?

astechow, 2020-05-27:
the pressure signal is what I meant - at the moment, a puff only has 1-2 data points in the scrolling display which is fine as an overview. As a "post-puff" view, a static display of the two pressure signal (+derivate or whatever might be useful here) would be nice

sballin, 2020-05-27:
Hmm doesn't this happen already [here](https://github.com/sballin/GPI_W7-X/blob/master/gui.py#L622)?

closed 2021-03-31 by sballin