sballin/GPI_W7-X #13: gauge displays value close to 0 when disconnected, potentially dangerous for pump

astechow, 2020-05-26:
The absolute pressure gauge controller displays values around 1 Torr when disconnected from the gauge. If "pumpdown and refill" is chosen in this state, the valve to the pump will open, potentially damaging it. Is there a way to detect a malfunction of the controller?

jameslterry, 2020-05-26:
Is the pressure reading always within some small range around 1 Torr under this disconnected condition? If so, then perhaps we could do a check for this constant pressure reading being sustained for (say ) 3 seconds before starting or not starting the "pumpdown and refill" sequence.

astechow, 2020-05-27:
I would need to check this. I don't know if this depends on the calibration of the gauge as well which is done on the controller if I recall correctly. It definitely fluctuates, but maybe to a much smaller degree than when connected to the gauge