sballin/GPI_W7-X #9: bug: p asymptotically goes to negative value on pumping, valve closes early

astechow, 2020-05-26:
probably due to calibration issues, the pressure drops to a negative value and continues to asymptotically approach some negative value. The check for "is pumped out" should reflect this by leaving more time (waiting for values to change less than a threshold in a given time window?) for the pump to do its job.

jameslterry, 2020-05-27:
think that we did put a delay into the software, but waiting for a change less than a threshold ANDed to a value less then the "pumped-out" threshold is a good idea.

sballin, 2021-04-01:
After some testing, I think the combination of waiting for a certain threshold and change, along with the option to pump down to a non-zero pressure or zero, creates complicated edge cases that would require way too many if/else clauses to solve. To keep things simple, I recommend not changing the simple threshold behavior. The original problem can be solved by changing the PUMPED_OUT value to a negative one.

closed 2021-04-01 by sballin