sballin/GPI_W7-X #18: Red Pitaya connection gets messed up after a couple hours

sballin, 2020-12-15:
## Summary

After a couple hours of streaming data from the Red Pitaya over the network, an attempt to receive data from the socket hangs and we can't reconnect to the RP unless we restart the RP uWSGI server or RP itself.

## Background

(This stuff is on the [dev](https://github.com/sballin/GPI_W7-X/tree/dev) branch.)

Overview of setup:

RP FPGA <-> RP C++ <->  RP uWSGI server <-> middle_server.py (<-> gui.py which is not part of the problem)

The data of interest here are pressure gauge recordings. These get stored in a FIFO queue on the FPGA:

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/koheron-sdk/instruments/GPI_RP/GPI_RP.hpp#L192

then they are popped and put into a FIFO queue in C++:

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/koheron-sdk/instruments/GPI_RP/GPI_RP.hpp#L194

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/koheron-sdk/instruments/GPI_RP/GPI_RP.hpp#L251-L254

then they are popped into a vector when middle_server.py asks for them:

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/koheron-sdk/instruments/GPI_RP/GPI_RP.hpp#L195

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/koheron-sdk/instruments/GPI_RP/GPI_RP.hpp#L170-L180

This is how middle_server.py gets the data:

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/middle_server.py#L268

https://github.com/sballin/GPI_W7-X/blob/d886eda51908d13e57239477f0849c878e4742c6/koheron-sdk/instruments/GPI_RP/GPI_RP.py#L40-L42

https://github.com/sballin/GPI_W7-X/blob/9acbb0ad0a5652b848fe3f5a2873b55b671c4732/koheron-sdk/python/koheron/koheron.py#L412-L418

https://github.com/sballin/GPI_W7-X/blob/9acbb0ad0a5652b848fe3f5a2873b55b671c4732/koheron-sdk/python/koheron/koheron.py#L365-L368

which culminates in two calls to this method. The first is to get the the length of the vector, the second is to get the data in the vector:

https://github.com/sballin/GPI_W7-X/blob/9acbb0ad0a5652b848fe3f5a2873b55b671c4732/koheron-sdk/python/koheron/koheron.py#L350-L363

## Problem

After a couple hours of good operation, this problem randomly strikes. The first of the usual pair of calls to `recv_all` (which wants to receive 12 bytes of data that include the length of the vector) receives only values of `b''` from the RP, which don't count toward the 12 bytes middle_server.py is waiting for, and because `b'' != ''`, we don't break out of the loop and this can cause a memory leak in middle_server.py (we are using python3—this particular piece of code may have been designed for python2 where `b''==''`). If you restart middle_server.py, you get the following error:

```
Traceback (most recent call last):
  File "/home/sballinger/wg/GPI_W7-X_newArch/koheron/koheron.py", line 246, in __init__
    self.sock.connect((host, port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "middle_server.py", line 485, in <module>
    rp = RPServer()
  File "middle_server.py", line 127, in __init__
    rpConnection = koheron.connect(RP_HOSTNAME, name='GPI_RP')
  File "/home/sballinger/wg/GPI_W7-X_newArch/koheron/koheron.py", line 54, in connect
    client = KoheronClient(host)
  File "/home/sballinger/wg/GPI_W7-X_newArch/koheron/koheron.py", line 249, in __init__
    raise ConnectionError('Failed to connect to {}:{} : {}'.format(host, port, e))
requests.exceptions.ConnectionError: Failed to connect to rp3:36000 : [Errno 111] Connection refused
```

which I've only been able to fix by restarting the RP or doing `killall uwsgi` on it. But during normal operation we want to avoid SSHing into the RP.

## Make it fail faster

This script can be used to produce the "connection refused" error faster, though the behavior may not be exactly the same.

```
import koheron
import struct
koheron.connect('rp3', name='GPI_RP')
c = koheron.KoheronClient('rp3')

while True:
    c.send_command(2,32,[])
    reserved, class_id, func_id, length = struct.unpack('>IHHI', c.recv_all(struct.calcsize('>IHHI')))
    print(length)
```

## Experiments

* Doesn't happen when trying to get other data from the RP, just this vector seems to be cursed
* Doesn't happen when running middle_server.py (or the failure-causing script) directly on the RP
* Even when the RP refuses the connection attempt from middle_server.py, the RP uWSGI server logs this connection request. It's only when middle_server is trying to connect to the socket that the error happens.
* If you fix the `recv_all` method to break after receiving `b''`, the connection is still refused after restarting middle_server.py

zack-vii, 2020-12-15:
better way for checking an empty string is
```python
if not chunk:
    break
```

socket recv only returns an empty string if the writing end on the server was closed. do you contron the server side. gdb run it in a terminal and check why it it hanging up, and when it dies why it is refusing to accept new connections. look like a missing exception in the sender loop tha causes the entire thread to terminate.

sballin, 2020-12-18:
Should be fixed by https://github.com/sballin/GPI_W7-X/commit/b3e4e047d122afd2f4f123e33a3ef2f90c2353b6

closed 2020-12-18 by sballin